Avamander/arduino-tvout #94: Incorrect files

ChristopherCramer, 2015-12-04:
Every time I download the zip folder the files do not have the same name as they are posted online. TVout.h is titled TVout TVout.cpp is also titled TVout, and fontALL.h is titled fontALL; so all the examples which include TVout.h, and fontALL.h fail to compile. Even if I change the code to #include <TVout>, and #include <fontALL>, or just rename the files correctly in the libraries, the sketch still fails to compile. Is there any way I could fix this from my PC, or could you just change the zip folder?

Avamander, 2015-12-04:
You downloaded something somehow wrong then. These files have exactly the
same names as the versions online. Redownload and install the library
correctly.

abcbarryn, 2015-12-04:
I believe you are using windows with the default option turned on that hides file extensions. Try this procedure:

Displaying the File Extension in Windows Vista and Windows 7
Click the Start menu. ...
Type "folder options" (without the quotes). ...
A dialog box with the title "Folder Options" will appear. ...
Click to uncheck the box for "Hide extensions for known file types".
Click the "OK" button at the bottom of the dialog box.

Additionally you need to copy the TVout and TVoutfonts folders to the Arduino Libraries folder.

closed 2015-12-05 by Avamander