Avamander/arduino-tvout #95: Compile Error

lgcproductions, 2016-02-29:
Just installed the library and tried compiling the PAL demo. It's coming up with an error saying the fontsALL.h file is missing. Have checked and it's in the library folder.

Also Sync is on pin 8 for a cloned UNO.

Avamander, 2016-02-29:
You have failed to install the library correctly. Both font and tvout library have to be directly in the libraries folder. 

If it's a UNO the sync is where UNO's sync is, clones are identical in pinout.

closed 2016-02-29 by Avamander

lgcproductions, 2016-02-29:
So how do I install it correctly then? Ive been doing it via the arduino IDE. Sketch - Include library - Add library .zip.

I got the screen working on the pinout I described  with the R5.91 version of the library.

Avamander, 2016-02-29:
Open your file explorer and just copy the two libraries (fonts and tvout) from the .zip file into your libraries folder.

lgcproductions, 2016-02-29:
Ok done that and now i'm getting this error. Just trying to compile the PAL demo example.

https://dl.dropboxusercontent.com/u/4764626/tvout.PNG

Avamander, 2016-02-29:
You messed up your install somehow. Wipe it and try again.

lgcproductions, 2016-02-29:
Is that arduino IDE installation or the library? As all the other Library's ive installed work perfectly.

abcbarryn, 2016-02-29:
What was the first error encountered? What version of the Arduino IDE are you using? Is it the latest version? Please make sure that the TVoutfonts folder is copied up to the library folder, and make sure you are running the current version of the Arduino IDE.

lgcproductions, 2016-02-29:
Initially it would not compile the newest version of your library saying there was no fontALL.h file found. 
Running IDE 1.6.5. Yup certainly in my library folder as can be seen in the screen grab.

https://dl.dropboxusercontent.com/u/4764626/tvout2.PNG

xznhj8129, 2024-04-07:
What a mess