Avamander/arduino-tvout #96: tvout libriery dos not work on fatshark rcv922

veiland, 2016-06-19:
I have a problem with my fatshark rcv922 base. As you know this goggles are without receiver, so i decided to make this project. Everything works great on 4" monitor and on TV, but when i connect arduino uno to my goggles with RCA connector, cube appear one second and screen is freeze, in other second everything become bright and than dark. I tried different resistors but problem is the same. I check the video signal from other sources(camcorder, tv out, other video receivers). Something is not the same as in other video signals. Maybe arduino 328 controller frequency is slow? as i know there is a TVP5150a video decoder in a fatshark.

here is a video of my problem:
https://www.youtube.com/watch?v=8TZUDGM92wE&feature=youtu.be

Maybe I can use some IC to convert arduino analog video signal to normal TV signal

Please help

Avamander, 2016-06-19:
Without a oscilloscope you can't really see the difference in signals. But
I'd say the only way you are going to get a better signal is to use a
dedicated chip and those are really expensive (even more so now).

veiland, 2016-06-19:
Yes i looked at oscilloscope the real signal is somewhere to  3.15khz and on arduino there is only 1,2khz. What dedicated chip you can suggest?

veiland, 2016-06-19:
I think there must be something between arduino and fatshark. Some chip that do the job.

closed 2016-07-10 by Avamander