Avamander/arduino-tvout #97: TVout blocks serial?

travisfarmer, 2016-07-09:
[mod_tvout.pdf](https://github.com/Avamander/arduino-tvout/files/355759/mod_tvout.pdf)
I am using TVout on an UNO, and I have it display text that comes in on the hardware serial port. the problem is, the serial data only displays a random selection of chars from the serial port.
I am using the Arduino IDE 1.6.9
The attached file is my project code.

Avamander, 2016-07-10:
It does a lot of things, but you have to test with just the base sketch, open the serial, test if it works, send a few characters, test if it works and so on. I can't debug it for you, as far as I remember everything worked fine.

travisfarmer, 2016-07-10:
I tried the NTSCserialTerm sketch in the examples folder, and got the following errors. note i have not changed anything in that file.
`In file included from C:\Users\Travis\Documents\Backup\Arduino\libraries\arduino-tvout\examples\NTSCserialTerm\NTSCserialTerm.ino:2:0:

C:\Users\Travis\Documents\Backup\Arduino\libraries\pollserial/pollserial.h:46:16: error: conflicting return type specified for 'virtual void pollserial::write(uint8_t)'

   virtual void write(uint8_t);

                ^

In file included from C:\Users\Travis\Documents\Backup\arduino-1.6.9\hardware\arduino\avr\cores\arduino/Stream.h:26:0,

                 from C:\Users\Travis\Documents\Backup\arduino-1.6.9\hardware\arduino\avr\cores\arduino/HardwareSerial.h:29,

                 from C:\Users\Travis\Documents\Backup\arduino-1.6.9\hardware\arduino\avr\cores\arduino/Arduino.h:224,

                 from sketch\NTSCserialTerm.ino.cpp:1:

C:\Users\Travis\Documents\Backup\arduino-1.6.9\hardware\arduino\avr\cores\arduino/Print.h:48:20: error:   overriding 'virtual size_t Print::write(uint8_t)'

     virtual size_t write(uint8_t) = 0;

                    ^

exit status 1
Error compiling for board Arduino/Genuino Uno.`

The TVout component works, but when i add serial, i get the problems. i would use the pollserial method, but as shown, i get the above errors.

travisfarmer, 2016-07-11:
[http://brainwagon.org/2012/01/14/the-downside-of-arduino/](url)

Seems like this is a common problem with your library. I wish i had known before i soldered up my module. it's a neat idea, but thanks anyway.

closed 2016-07-11 by travisfarmer

Avamander, 2016-07-11:
I have to repeat, not my library, I am just re-hosting the library to make
it available from Arduino IDE. Also, you should always try the things you
wish to implement on a breadboard before soldering it together.

travisfarmer, 2016-07-16:
if your interested, i have a modified version on my GitHub that compiles for me.
[https://github.com/travisfarmer/mod_tvout2/tree/Old-TVout/Old-TVout](url) the mod_tvout.ino shows how i implemented it in my code. it compiles, and i have it uploaded on a standalone ATmega328PU.
As i said, if you are interested.
BTW, i didn't modify the Pollserial Library. that was done by /dev from the Arduino forum.

Sorry for my earlier comment.

fredericplante, 2016-10-02:
Do like this

```
#include <TVout.h>
#include <fontALL.h>
#include <SoftwareSerial.h>

SoftwareSerial mySerial(10, 11); // RX, TX

TVout TV;

void setup()  {
  TV.begin(_NTSC,184,72);
  TV.select_font(font6x8);
  TV.println("Serial Terminal");
  TV.println("-- Version 0.1 --");
  //TV.set_hbi_hook(mySerial);
  mySerial.begin(4800);
  //mySerial.println("Hello, world?");
}

void loop() {
  if (mySerial.available()) {
    TV.print((char)mySerial.read());
  }
}
```

> Sketch uses 9,064 bytes (29%) of program storage space. Maximum is 30,592 bytes.
> Global variables use 205 bytes of dynamic memory.

on a gamebuino bootloader.