Avamander/arduino-tvout #100: Screen rolling with USB2 capture card.

morrisonj, 2017-01-15:
I'm using one of these cheap USB2 capture cards: http://www.banggood.com/USB2_0-Video-TV-Tuner-DVD-Audio-Capture-Card-Converter-Record-Receiver-p-921000.html

I get screen rolling/scrolling with the default settings. When I change _PAL_LINE_FRAME from 312 to 320 in spec/video_properties.h the rolling/scrolling goes away and the problem is fixed.

Maybe this might help others with a similar issue.

Avamander, 2017-01-15:
Thank you for your information. Hopefully people who need this will find it when googling.

closed 2017-01-15 by Avamander

zoomx, 2017-01-24:
The original library had this issue on PAL version.
The solution was in a issue on the Google Code repository.
It was in video_gen.cpp in vsync_line function
`void vsync_line() {

	if (display.scanLine >= display.lines_frame) {

		OCR1A = _CYCLES_VIRT_SYNC;
		display.scanLine = 0;
		//https://code.google.com/p/arduino-tvout/issues/detail?id=7`
were display.scanLine = 0; was changed to display.scanLine = -1;

Maybe it is still useful.

morrisonj, 2017-01-24:
Thanks, I did come across that but changing display.scanLine didn't make any difference in my case.