Avamander/arduino-tvout #106: Not compiling examples on Arduino 1.8.2

ricardojlrufino, 2017-07-05:
The examples do not compile giving a linkage error. I had to make the following changes to work.

Rename folder **TVoutfonts** to **utility**
Remove folder TVout (Has a folder inside the other)
Change import to #include <utility/fontALL.h> in sample

hafo821, 2017-07-07:
Thank´s for this comment... I spent the whole hour of trying to compile these f;king samples

allnewmike, 2017-07-10:
Me too!

Wish I'd have looked here first.

nathantsoi, 2017-07-10:
@ricardojlrufino, can you make a PR so we can all use your changes ?

ric96, 2017-07-10:
@nathantsoi https://github.com/Avamander/arduino-tvout/pull/107

treaki, 2017-09-08:
i had a similar issue and i solved it by just coping all files from ~/Arduino/libraries/TVout/TVoutfonts to Arduino/libraries/TVout ... Then it worked

closed 2020-06-06 by Avamander