Avamander/arduino-tvout #112: I have some fixes for TVOut

spaceneedle, 2017-12-18:
Hello, i tried to email avamander, but it bounced. 

As you may or may not know, TVOut is broken these days in the latest Arduino release due to an issue the way assembler macros are handled by the compiler. The hilarious thing is that pops up intermittently depending on what TVOut functions are placed next to what. This is the famous "opcode error" that spews out while compiling. I have a fix for it, but I am wondering if you can add me as a contributor and create a "development" branch so we can get this fix out?

I also have a fix for pollserial, which is also broken in the current release. It also improves screen stability while using pollserial. (apparently screen instability popped up a few Arduino releases ago as well).

Avamander, 2017-12-18:
Not sure what e-mail you got... But anyways, if you have fixes, submit a pull request.

spaceneedle, 2017-12-18:
Looks like i typoed it .. i'll submit a pull request later today when I get home. Thanks!

PatrickWeaver, 2018-02-11:
@spaceneedle Did these changes ever get merged in?

johnbentcope, 2018-04-14:
@PatrickWeaver It looks like they have not. Last commit was April 2017, @spaceneedle posted this in December 2017.

zoomx, 2018-04-16:
It seems that there isn't any pull request by spaceneedle

tomsykes, 2018-07-09:
Hi @spaceneedle - if you could share your changes, I can help you submit a pull request, or do it for you.

tomsykes, 2018-07-09:
I am, but I admit I'm using 1.6.12. I'll try updating, but the problem is rather random in its nature.

Godfear17, 2018-09-02:
cannot get it to work in 1.8.5...

Avamander, 2018-09-02:
@Godfear17 https://github.com/Avamander/arduino-tvout/issues/106#issuecomment-328036536

closed 2018-09-02 by Avamander

spaceneedle, 2019-08-22:
Hey guys. I have picked up my TVOut project again and i just noticed this. I feel bad abandoning everyone. I have the fix ready for everyone, but for whatever reason, github is not allowing me to submit a pull request to this repo. This library makes use of some assembly macros that break due to something really stupid (I can't remember what triggers it, but it seems that once you get a good project going, it effectively has issues). @Avamander I will send you an email directly with this fix.

Also, I noticed some issues with the fonts and the examples. It seems like the Arduino IDE can't find them. Either the include is not doing a full search or something else is an issue. Moving them to the root directory fixed the issue.

As I mentioned earlier regarding my project, I am launching a TVOut kit complete with cart slot and two atari compatible joystick ports in September 2019. The main board is game capable, but when a cart is inserted, it relinquishes the video output to the cartridge and relays game controller input (both digital and analog atari joysticks) via serial. Should be a lot of fun!