Avamander/arduino-tvout #118: Work on TV and some small monitor

Ratthanin, 2019-07-16:
Work very well on most screen but just curious why it doesn't work on some small screen .
for example.  4.3" or 3.5" rear camera monitor.  the monitor works well with Raspberry Pi composite video but not Arduino.
I have measure the voltage is around 1.4 - 1.6 V  almost the same. but no picture.

Just curious  why?   
Thanks

zoomx, 2019-07-16:
I have one of these small monitor and it works.
Take in account that maybe these monitor works better in NTSC than PAL or maybe they work only in NTSC or PAL.

Ratthanin, 2019-07-20:
T

> I have one of these small monitor and it works.
> Take in account that maybe these monitor works better in NTSC than PAL or maybe they work only in NTSC or PAL.

thank you,
I also notice that if I draw rectangle or bitmap outside the screen boundary
for example the screen resolution is 128x96 pixels, and my code is TV.bitmap(0,-1,mybitmap);
Arduino will reset somehow.
It's like coordinate X and Y must not be less than 0.
Also arduino nano with old bootloader will stuck in watchdog reset loop, keep reset forever.  I have to uplaod sketch to make it work again.
Should be good to add code to check minus poisition and avoid crash.

closed 2020-06-06 by Avamander