Avamander/arduino-tvout #123: It's a important assembler BUG.

LiuTianyouOnLuogu, 2020-08-06:
`C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s: Assembler messages:

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:194: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:233: Error: unknown opcode `delay1'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:235: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:238: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:241: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:244: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:247: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:250: Error: unknown opcode `delay1'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:255: Error: unknown opcode `delay1'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:258: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:291: Error: unknown opcode `svprt'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:295: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:299: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:300: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:302: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:303: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:305: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:306: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:308: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:309: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:311: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:312: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:314: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:315: Error: unknown opcode `delay1'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:318: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:320: Error: unknown opcode `delay1'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:322: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:323: Error: unknown opcode `svprt'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:325: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:357: Error: unknown opcode `svprt'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:361: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:364: Error: unknown opcode `delay1'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:366: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:367: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:369: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:370: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:372: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:373: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:375: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:376: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:378: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:379: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:381: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:382: Error: unknown opcode `delay3'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:384: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:387: Error: unknown opcode `delay2'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:389: Error: unknown opcode `o1bs'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:390: Error: unknown opcode `svprt'

C:\Users\13472\AppData\Local\Temp\ccGw4Y82.s:392: Error: unknown opcode `o1bs'

lto-wrapper: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status`

LiuTianyouOnLuogu, 2020-08-06:
HOW CAN I DO FOR IT!!!

Avamander, 2020-08-06:
Use a supported microprocessor.

closed 2020-08-06 by Avamander

LiuTianyouOnLuogu, 2020-08-06:
What's the supported microprocessor?
Does Arduino UNO have it?

LiuTianyouOnLuogu, 2020-08-06:
You're SO fast.

LiuTianyouOnLuogu, 2020-08-06:
How can I use it at Arduino UNO?

Avamander, 2020-08-06:
The library currently works on ATmega168, ATmega328, ATmega1280, ATmega2560, ATmega644p, ATmega1284p, ATmega32U4 and AT90USB1286.

LiuTianyouOnLuogu, 2020-08-07:
It's same as ATmega2560.

LiuTianyouOnLuogu, 2020-08-07:
What's the difference about ATmega1280 amd ATmega128P?
I think it's a bug.

LiuTianyouOnLuogu, 2020-08-07:
Of cause, I get the same error at ATmega2560 and ATmega1280.

LiuTianyouOnLuogu, 2020-08-07:
I tried to compile it at all kinds of  microprocessors.

LiuTianyouOnLuogu, 2020-08-07:
Excuse me for interrupting you.
But in China, It's 2020/8/7 10:36 now.

Avamander, 2020-08-08:
Try an older version of the IDE or disable LTO.