Avamander/arduino-tvout #125: Do arduino-tvout not support the type "const __FlashStringHelper*"?

LiuTianyouOnLuogu, 2020-08-11:
Compiler gives me a error "call of overloaded 'print(const __FlashStringHelper*)' is ambiguous" when I am using "TV.print(F("hello, world"))".

Why come?

I think this function is only declare and it is not defined.

LiuTianyouOnLuogu, 2020-08-11:
This is only error:
```C:\Users\13472\AppData\Local\Temp\buildc463139c3f218ebfbc5fbbbd280d322c.tmp\sketch\LiuTianyou-Basic.cpp: In member function 'void Basic::setup()':

LiuTianyou-Basic.cpp:24: error: call of overloaded 'print(const __FlashStringHelper*)' is ambiguous

   out.print(F("Initializing SD card..."));

                                         ^

C:\Users\13472\AppData\Local\Temp\buildc463139c3f218ebfbc5fbbbd280d322c.tmp\sketch\LiuTianyou-Basic.cpp:24:41: note: candidates are:

In file included from C:\Users\13472\AppData\Local\Temp\buildc463139c3f218ebfbc5fbbbd280d322c.tmp\sketch\header.h:61:0,

                 from C:\Users\13472\AppData\Local\Temp\buildc463139c3f218ebfbc5fbbbd280d322c.tmp\sketch\LiuTianyou-Basic.cpp:1:

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:127:10: note: void TVout::print(char, int) <near match>

     void print(char, int = BYTE);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:127:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'char'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:128:10: note: void TVout::print(unsigned char, int) <near match>

     void print(unsigned char, int = BYTE);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:128:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'unsigned char'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:129:10: note: void TVout::print(int, int) <near match>

     void print(int, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:129:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'int'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:130:10: note: void TVout::print(unsigned int, int) <near match>

     void print(unsigned int, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:130:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'unsigned int'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:131:10: note: void TVout::print(long int, int) <near match>

     void print(long, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:131:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'long int'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:132:10: note: void TVout::print(long unsigned int, int) <near match>

     void print(unsigned long, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:132:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'long unsigned int'

LiuTianyou-Basic.cpp:29: error: call of overloaded 'println(const __FlashStringHelper*)' is ambiguous

   out.println(F("LiuTianyou-Basic For Arduino Operator System"));

                                                                ^

C:\Users\13472\AppData\Local\Temp\buildc463139c3f218ebfbc5fbbbd280d322c.tmp\sketch\LiuTianyou-Basic.cpp:29:64: note: candidates are:
[... 485 lines cut ...]
C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:157:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'int'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:158:10: note: void TVout::println(unsigned int, int) <near match>

     void println(unsigned int, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:158:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'unsigned int'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:159:10: note: void TVout::println(long int, int) <near match>

     void println(long, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:159:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'long int'

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:160:10: note: void TVout::println(long unsigned int, int) <near match>

     void println(unsigned long, int = DEC);

          ^

C:\Users\13472\Documents\Arduino\libraries\TVout/TVout.h:160:10: note:   no known conversion for argument 1 from 'const __FlashStringHelper*' to 'long unsigned int```

LiuTianyouOnLuogu, 2020-08-11:
I only replace Serial and TVout.
Look at here:
```
#ifdef computer-like
#include <TVout.h>
class output: public TVout {
  public:
    output() {
      begin(PAL);
    }
    /*print(const __FlashStringHelper* x){
      print(String((const __FlashStringHelper*)x).c_str());
    }
    println(const __FlashStringHelper* x){
      println(String((const __FlashStringHelper*)x).c_str());
    }*/
#else
class output: public HardwareSerial {
  public:
    output() {
      Serial.begin(115200);
    }
#endif
};
```

LiuTianyouOnLuogu, 2020-08-11:
I think it supported.
But why comes?
I don't know why.

Avamander, 2020-08-11:
You probably need to use an older Arduino IDE version.

LiuTianyouOnLuogu, 2020-08-12:
Why older Arduino IDE version?
I think you need to fix this problem to support newer Arduino IDE and C++ 11.
You know, some libraries need newer IDE.

LiuTianyouOnLuogu, 2020-08-12:
What is the difference between newer and older Arduino IDE?

LiuTianyouOnLuogu, 2020-08-12:
So, what version IDE I need?

LiuTianyouOnLuogu, 2020-08-12:
And you know, Chinese can't visit Google Code.(Great Firewall)

LiuTianyouOnLuogu, 2020-08-13:
look this at TVout.h:
```
void write(uint8_t);
    void write(const char *str);
    void write(const uint8_t *buffer, uint8_t size);

    void print(const char[]);
    void print(char, int = BYTE);
    void print(unsigned char, int = BYTE);
    void print(int, int = DEC);
    void print(unsigned int, int = DEC);
    void print(long, int = DEC);
    void print(unsigned long, int = DEC);
    void print(double, int = 2);
	
	void print(uint8_t, uint8_t, const char[]);
	void print(uint8_t, uint8_t, char, int = BYTE);
	void print(uint8_t, uint8_t, unsigned char, int = BYTE);
	void print(uint8_t, uint8_t, int, int = DEC);
	void print(uint8_t, uint8_t, unsigned int, int = DEC);
	void print(uint8_t, uint8_t, long, int = DEC);
	void print(uint8_t, uint8_t, unsigned long, int = DEC);
	void print(uint8_t, uint8_t, double, int = 2);
	
	void println(uint8_t, uint8_t, const char[]);
    void println(uint8_t, uint8_t, char, int = BYTE);
    void println(uint8_t, uint8_t, unsigned char, int = BYTE);
    void println(uint8_t, uint8_t, int, int = DEC);
    void println(uint8_t, uint8_t, unsigned int, int = DEC);
    void println(uint8_t, uint8_t, long, int = DEC);
    void println(uint8_t, uint8_t, unsigned long, int = DEC);
    void println(uint8_t, uint8_t, double, int = 2);
    void println(uint8_t, uint8_t);

    void println(const char[]);
    void println(char, int = BYTE);
    void println(unsigned char, int = BYTE);
    void println(int, int = DEC);
    void println(unsigned int, int = DEC);
    void println(long, int = DEC);
    void println(unsigned long, int = DEC);
    void println(double, int = 2);
    void println(void);
```
It has too many overload, and many overload is unneed and unuseful.
And please look there:
```
void TVout::print(const char str[])
{
  write(str);
}

void TVout::print(char c, int base)
{
  print((long) c, base);
}

void TVout::print(unsigned char b, int base)
{
  print((unsigned long) b, base);
}

void TVout::print(int n, int base)
{
  print((long) n, base);
}

void TVout::print(unsigned int n, int base)
{
  print((unsigned long) n, base);
}

void TVout::print(long n, int base)
{
  if (base == 0) {
    write(n);
  } else if (base == 10) {
    if (n < 0) {
      print('-');
      n = -n;
    }
    printNumber(n, 10);
  } else {
    printNumber(n, base);
  }
}

void TVout::print(unsigned long n, int base)
{
  if (base == 0) write(n);
  else printNumber(n, base);
}

void TVout::print(double n, int digits)
{
  printFloat(n, digits);
}

void TVout::println(void)
{
  print('\r');
  print('\n');  
}

void TVout::println(const char c[])
{
  print(c);
  println();
}

void TVout::println(char c, int base)
{
  print(c, base);
  println();
}

void TVout::println(unsigned char b, int base)
{
  print(b, base);
  println();
}

void TVout::println(int n, int base)
{
  print(n, base);
  println();
}

void TVout::println(unsigned int n, int base)
{
  print(n, base);
  println();
}

void TVout::println(long n, int base)
{
  print(n, base);
  println();
}

void TVout::println(unsigned long n, int base)
{
  print(n, base);
  println();
}

void TVout::println(double n, int digits)
{
  print(n, digits);
  println();
}
```
It has many repats, and no const __FlashStringHelper* type.

LiuTianyouOnLuogu, 2020-08-13:
only 
```
void TVout::print(const char str[]);
void TVout::print(int n, int base);
void TVout::println(void);
void TVout::println(const char c[]);
void TVout::println(int n, int base);
```
is needed, another types can be casted.

LiuTianyouOnLuogu, 2020-08-13:
@Avamander these functions you need:
```
print(const __FlashStringHelper* x){
      print(String((const __FlashStringHelper*)x).c_str());
}
println(const __FlashStringHelper* x){
      println(String((const __FlashStringHelper*)x).c_str());
}
```

LiuTianyouOnLuogu, 2020-08-13:
And this:
```
print(String x){
      print(x.c_str());
}
println(String x){
      print(x.c_str());
}
```

Avamander, 2020-08-13:
> So, what version IDE I need?

Legacy one, below 1.5 most likely.

LiuTianyouOnLuogu, 2020-08-13:
You know, China can't see some pages.
So why you can't add the support for new IDE?

LiuTianyouOnLuogu, 2020-08-13:
I will try to fix it.

LiuTianyouOnLuogu, 2020-08-13:
God, I think I can't.
It's have too many overload and unuseful code!

closed 2022-01-20 by LiuTianyouOnLuogu