Avamander/arduino-tvout #134: A Note For This Project

LiuTianyouOnLuogu, 2022-01-20:
I send an issue about I can't accept compiling in latest Arduino version. However, now I solve this problem!

You need use Arduino IDE v1.6.6 and the latest TVout version in Github.

You may change the place of TVfontsALL.h so that you can use fonts.

I'm so suprised that I solve this! And I think nobody will make the same mistake again.

@Avamander

LiuTianyouOnLuogu, 2022-01-20:
It's a long time. Today I found this issue and I solve this.

Avamander, 2022-01-20:
Yep, this is because there's technically two libraries in one, it's a remnant of what it was ten years ago.

LiuTianyouOnLuogu, 2022-01-20:
Solution of #124

CodeAsm, 2022-03-31:
> Yep, this is because there's technically two libraries in one, it's a remnant of what it was ten years ago.

It might help combine these 2 libs into one or make mention ofthis somewhere more insight ? Im happy this issue was "easy" for me to find.

closed 2022-05-15 by Avamander