Avamander/arduino-tvout #138: a font bug need fix or notice

LiuTianyouOnLuogu, 2022-08-04:
In font files, i found this:
```cpp
//[
0b11000000,
0b10000000,
0b10000000,
0b10000000,
0b11000000,
0b00000000,
0,				//for the life of me I have no idea why this is needed....
//\
0b10000000,
0b10000000,
0b01000000,
0b00100000,
0b00100000,
0b00000000,
```

If you use `//\`, it will let the comment extend to the next line, so it is useless for the line of `0b10000000,`.

So in fact the charcacter '\' is:

```cpp
0
0b10000000,
0b01000000,
0b00100000,
0b00100000,
0b00000000,
```
It's a bug.

LiuTianyouOnLuogu, 2022-08-04:
In file font4x6.cpp.

LiuTianyouOnLuogu, 2022-08-04:
I fixed it with

```cpp
//[
0b11000000,
0b10000000,
0b10000000,
0b10000000,
0b11000000,
0b00000000,
//(backslash)
0b10000000,
0b10000000,
0b01000000,
0b00100000,
0b00100000,
0b00000000,
```

LiuTianyouOnLuogu, 2022-08-04:
@Avamander

LiuTianyouOnLuogu, 2022-08-08:
Well, if you open -Wall & -Wextra, you'll get:

```
thirdparty/TVout/TVoutfonts/font4x6.cpp:426:1: warning: multi-line comment [-Wcomment]
 //\
 ^
```

orangefabri, 2022-10-20:
I remember I found same bug on the code. 

did you made font the size is not the multiple of 8?

I wanted to make 11x11 font but it was very difficult.

8 or 16 font was easy to made

LiuTianyouOnLuogu, 2022-10-23:
> I remember I found same bug on the code.
> 
> did you made font the size is not the multiple of 8?
> 
> I wanted to make 11x11 font but it was very difficult.
> 
> 8 or 16 font was easy to made

Well, if the size is not the multple of 8, it will be very diffcult to make font allign on the screen.

And the code are running base on the multiple of 8. The wide of the screen is multiple of 8.

So That's diffcult. I just want to notice everyone there's a clearly bug on the font.

LiuTianyouOnLuogu, 2022-10-23:
@orangefabri If you are interested in the code, why not join the develop?