Avamander/arduino-tvout #149: NTSCserialTerm compile error

OwensDataCenter, 2023-09-27:
`Arduino: 1.8.19 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from C:\Users\Workshop\Documents\Arduino\libraries\TVout\examples\NTSCserialTerm\NTSCserialTerm.pde:2:0:

C:\Users\Workshop\Documents\Arduino\libraries\pollserial/pollserial.h:46:16: error: conflicting return type specified for 'virtual void pollserial::write(uint8_t)'

   virtual void write(uint8_t);

                ^~~~~

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Stream.h:26:0,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/HardwareSerial.h:29,

                 from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:233,

                 from sketch\NTSCserialTerm.pde.cpp:1:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Print.h:51:20: error:   overriding 'virtual size_t Print::write(uint8_t)'

     virtual size_t write(uint8_t) = 0;

                    ^~~~~

exit status 1

Error compiling for board Arduino Mega or Mega 2560.`

radio-satellites, 2023-10-24:
I have the same issue here.

skaalit, 2024-06-30:
hello.

solved by changing line 131 in pollserial.cpp from
`void pollserial::write(uint8_t c) {
`
to 
`size_t pollserial::write(uint8_t c) {
`

and line 46 in pollserial.h from 
`virtual void write(uint8_t);`
to
`virtual size_t write(uint8_t);`

I know nothing about coding. but seems to compile and work. hope this helps :3

closed 2024-07-01 by damienwolf07